f9micro/f9-kernel #118: Think I just hosed a discovery board with default stm32f4 config

jeremytregunna, 2015-12-03:
Ran make config, selected the stm32f4 (have a 407vg board), didn't check the HSE_VALUE in platform/stm32f4 which is set to 1 << 27 i see now. Valid values should be 8000000 not 134217728. stlink no longer connects to the board anymore, and I can't find a way to get it to boot up. Just to be clear, I made no changes from what's in master.

Am I completely out of luck here, or just missing something?

jserv, 2015-12-03:
@jeremytregunna : Thanks for reporting! Can you try to use openocd to flash STM32F407 board?

jeremytregunna, 2015-12-03:
@jserv I did, and that did not work also. So out of desperation, I used ST's windows tool, which was able to detect the board, erase the flash, and load new code onto it. Now openocd recognizes it again.

closed 2015-12-08 by jeremytregunna