f9micro/f9-kernel #142: Is there some demo code using f9-kernel?

gnitnaw, 2017-02-18:
Like control of gpio or I2C device. I would like do some test.

gapry, 2017-02-19:
For now, it doesn't exist device driver of GPIO or I2C.

F9 microkernel is supported the following platforms.
1. STM32F4DISCOVERY
2. STM32F429
3. STM32P103

If your interesting platforms are not above, you need to reference the following implementations to know how to add a new platform in F9 microkernel.   

Reference: [Support Platform](https://github.com/f9micro/f9-kernel/tree/master/board)

For example, I assume you need to use GPIO in STM32F429. First, you need the device driver of STM32F429. You can reference the following implementation and article. 

0. [Article](https://stm32f4-discovery.net/2014/04/stm32f429-discovery-gpio-tutorial-with-onboard-leds-and-button/)
1. [Source Code](https://github.com/MaJerle/stm32f429/blob/master/53-STM32F429_GPIO/User/main.c)

Second, you need to refactor it which will be a user apps in F9 microkernel. Also, you need to reference the following implementations to know how to write a user app in F9 microkernel.  BTW, I strongly suggest you to see the pingpong app for the first time.

Reference: [User APPs](https://github.com/f9micro/f9-kernel/tree/master/user/apps)

gnitnaw, 2017-02-19:
Thanks. Hopefully I can also contribute this project.

closed 2017-02-19 by gnitnaw