f9micro/f9-kernel #143: userApp

gnitnaw, 2017-03-06:
- If I use UART4 to connect, then the screen will print lots of " � ", is it correct?
- How can I change userApp (for example, use "l4test" instead of "pingpong")?
- Can I run several userApp at the same time?
Thanks.

grapherd, 2017-03-07:
I think the answer is NO.

Could you give me the info about your board and the setting in the `user/apps/build.mk`?
Because if you are using STM32F429, and building with LCD apps, you may hang by this process, and got the junk info like you got.

gnitnaw, 2017-03-07:
I didn't change the code (with stm32f429 discovery). All are the same.
I connect to USART4: PC11 (TX), PC10 (RX) 
Here is the message after I push restart button:

====================================================
 Copyright(C) 2013-2014 The F9 Microkernel Project  
====================================================
Git head: d466fe1dcf8c0b230a6e7397ba4402bf5812bd37
Host: x86_64
Build:������������������������������������������������������������������������������������������������������������������������������������������������....(the same)

About LCD apps, I can see the rectangle on the LCD.

grapherd, 2017-03-07:
You still need to check `user/apps/build.mk`, my setting is like this
```
# Copyright (c) 2013 The F9 Microkernel Project. All rights reserved.
# Use of this source code is governed by a BSD-style license that can be
# found in the LICENSE file.

user-apps-dirs = \
	l4test \
```

If you use default setting, it may contain other apps.

And if I remember correctly, UART output will use async output, so your kernel may occure some problem after `__l4_printf(banner)` when init with `INIT_LEVEL_KERNEL_EARLY` hook at `start.c`.

Maybe you can use gdb to trace it.

gnitnaw, 2017-03-07:
After I remove the ping-pong in user/apps/build.mk, it becomes:

====================================================
 Copyright(C) 2013-2014 The F9 Microkernel Project  
====================================================
Git head: d466fe1dc�������������������������������������������������������������������������������������������������������������������������������

So how to solve this problem?

gapry, 2017-03-07:
What is your current baud rate? 115200 or 9600

gnitnaw, 2017-03-07:
I have tried 115200, 76800, 38400, 19200, and 9600.
Only 115200 can let me see the banner...

grapherd, 2017-03-07:
@gnitnaw  Can you give me your `.config`?

gnitnaw, 2017-03-07:
I did "make config" and select STM32F429 in the platform and save, that's all.
=======================
CONFIG_BOARD_STM32F429DISCOVERY=y
# CONFIG_BOARD_STM32P103 is not set
# CONFIG_BITMAP_BITBAND is not set
# CONFIG_FPU is not set
# CONFIG_STDIO_NODEV is not set
CONFIG_STDIO_USE_DBGPORT=y
# CONFIG_DBGPORT_USE_USART1 is not set
# CONFIG_DBGPORT_USE_USART2 is not set
CONFIG_DBGPORT_USE_USART4=y

#
# Limitations
#
CONFIG_MAX_THREADS=32
CONFIG_MAX_KT_EVENTS=64
CONFIG_MAX_ASYNC_EVENTS=32
CONFIG_MAX_ADRESS_SPACES=16
CONFIG_MAX_FPAGES=256

#
# Kernel Timer
#
CONFIG_KTIMER_TICKLESS=y
# CONFIG_KTIMER_TICKLESS_VERIFY is not set
CONFIG_KTIMER_TICKLESS_COMPENSATION=0
CONFIG_KTIMER_TICKLESS_INT_COMPENSATION=0
CONFIG_KTIMER_HEARTBEAT=65536
CONFIG_KTIMER_MINTICKS=128

#
# Flexible page tweaks
#
CONFIG_LARGEST_FPAGE_SHIFT=16
CONFIG_SMALLEST_FPAGE_SHIFT=8

#
# Thread tweaks
#
CONFIG_INTR_THREAD_MAX=256

#
# KIP tweaks
#
CONFIG_KIP_EXTRA_SIZE=128

#
# Kernel hacking
#

CONFIG_DEBUG=y
CONFIG_DEBUG_DEV_UART=y
CONFIG_KDB=y
CONFIG_KPROBES=y
CONFIG_SYMMAP=y
CONFIG_PANIC_DUMP_STACK=y
# CONFIG_LOADER is not set

#
# Test Case
#

gapry, 2017-03-07:
I assume you use minicom to see the output of UART. So, do you have turned
off (or turned on) software flow control and hardware flow control ?

gnitnaw, 2017-03-07:
No, I simply follow your instruction:
sudo screen /dev/ttyUSB0 115200 8n1

grapherd, 2017-03-07:
That's weird. Using your setting with `master` (d466fe1dcf8c) branch, I can get the output:

```bash
====================================================
 Copyright(C) 2013-2014 The F9 icrokernel Project  
====================================================
Git head: d466fe1dcf8c0b230a6e7397ba4402bf5812bd37
Host: x86_64
Build: 2017-03-07T18:32:35+0800

Press '?' to print KDB menu

L4/Pistachio test suite starts

Simple IPC test (inter-as, only untyped words)
  Send Message transfer:   OK
  ReplyWait Message transfer:   OK
Returned Id 0 != 0 (local) [0 (global)]
  From parameter (local):   FAILED
Kernel panic: Hard fault. Restarting
-------KTABLES------
```

gnitnaw, 2017-03-07:
If I use UART2 (change setting in .config , and connect to PD5/PD6) instead of UART4, I get:

====================================================
 Copyright(C) 2013-2014 The F9 Microkernel Project  
====================================================
Git head: d466fe1dcf8c0b230a6e7397ba4402bf5812bd37
Host: x86_64
 tt y sma2d o)0

## KDB ##

gnitnaw, 2017-03-07:
Here is the message with USART1:

 ====================================================
 Copyright(C) 2013-2014 The F9 Microkernel Project  
 ====================================================
Git head: d466fe1dcf8c0b230a6e7397ba4402bf5812bd37
Host: x86_64
Build: 2017-03-07T12:08:07+0100

Press '?' to print KDB menu

L4/Pistachio test suite starts

Simple IPC test (inter-as, only untyped words)
  Send Message transfer:   OK
  ReplyWait Message transfer:   OK
Returned Id 0 != 0 (local) [0 (global)]
  From parameter (local):   FAILED
lcd main

gapry, 2017-03-07:
For now, I know you have tried USART1, UART2 and UART4 but it doesn't work normally. 

Hence, I suggest you to read the document [STM32F420 Note](https://stm32f429.hackpad.com/NOTE-WbiooOfkaoR) at this moment, please pay attention [「示範專案：USART 測試程式」](https://stm32f429.hackpad.com/NOTE-WbiooOfkaoR#:h=%E7%A4%BA%E7%AF%84%E5%B0%88%E6%A1%88%EF%BC%9AUSART-%E6%B8%AC%E8%A9%A6%E7%A8%8B%E5%BC%8F), which is a part of the document.

If it doesn't work, I suggest you change your USB TTL Cable.

gnitnaw, 2017-03-07:
For me USART1 works...

ab1aw, 2019-03-05:
Are you also using the LEDs?  The LEDs and the USART4 (PA0 and PA1) are using the same GPIO pins.

jserv, 2026-01-28:
Close as workaround existed.