f9micro/f9-kernel #144: DECLARE_USER

gnitnaw, 2017-03-07:
I'm creating an app. for stm32f429 discovery.
Here is my code:
https://github.com/gnitnaw/f9-kernel/blob/master/user/apps/led/main.c

I would like to ask how to write the part "DECLARE_USER" and "DECLARE_FPAGE".

Another question: I have one stm32f072b and one stm32l476 board.
Are the files in platform/stm32-common compatible with these two boards?
Thanks.

gapry, 2017-03-07:
Can it work now?

gnitnaw, 2017-03-07:
No. I got lots of like this : 
MEM:   fpage APB2_3DEV [b:40015400, sz:2**10]
MEM:   fpage APB2_3DEV [b:40015800, sz:2**10]
MEM:   fpage APB2_4DEV [b:40016800, sz:2**11]
MEM:   fpage APB2_4DEV [b:40017000, sz:2**8]
MEM:   fpage APB2_4DEV [b:40017100, sz:2**8]
MEM:   fpage APB2_4DEV [b:40017200, sz:2**9]
MEM:   fpage AHB1_1DEV [b:40020000, sz:2**10]
MEM:   fpage AHB1_1DEV [b:40020400, sz:2**10]
MEM:   fpage AHB1_1DEV [b:40020800, sz:2**11]
MEM:   fpage AHB1_1DEV [b:40021000, sz:2**12]
MEM:   fpage AHB1_1DEV [b:40022000, sz:2**10]
MEM:   fpage AHB1_1DEV [b:40022400, sz:2**10]
MEM:   fpage AHB1_1DEV [b:40022800, sz:2**11]
MEM:   fpage AHB1_1DEV [b:40023000, sz:2**10]
MEM:   fpage AHB1_1DEV [b:40023400, sz:2**10]
MEM:   fpage AHB1_1DEV [b:40023800, sz:2**10]
MEM:   fpage AHB1_2DEV [b:40028000, sz:2**15]
MEM:   fpage CR_PLLSAION_BB [b:42470000, sz:2**10]
MEM:   fpage CR_PLLSAION_BB [b:42470400, sz:2**10]
MEM:   fpage CR_PLLSAION_BB [b:42470800, sz:2**10]
MEM:   fpage AHB3DEV [b:a0000000, sz:2**12]
MEM:   fpage LCD_FRAME_BUFFER_1 [b:d0000000, sz:2**17]
MEM:   fpage LCD_FRAME_BUFFER_1 [b:d0020000, sz:2**17]
MEM:   fpage LCD_FRAME_BUFFER_1 [b:d0040000, sz:2**18]
MEM:   fpage LCD_FRAME_BUFFER_1 [b:d0080000, sz:2**17]
MEM:   fpage LCD_FRAME_BUFFER_2 [b:d00a0000, sz:2**17]
MEM:   fpage LCD_FRAME_BUFFER_2 [b:d00c0000, sz:2**18]
MEM:   fpage LCD_FRAME_BUFFER_2 [b:d0100000, sz:2**18]
-------TOP------
Init sampling...

Stack dump:
20000388 00000082 2000e6f4 40023810 00000003 00000004 08001395 080051cc 
00000082 40023810 00408002 20013e80 200123ba 10000835 f0000001 00000003 
20012f2c 00000003 080013af fffffffd 00000001 08005854 00000000 0800439d 
40020024 0800401b 080043a4 61000000 0800439d 08004321 00000000 0800430d 
0000eb58 0800436d e000ed24 0800414d 00000000 ffffffff 00000000

gnitnaw, 2017-03-07:
If the code works well, I should wee the LD3 toggle...

gapry, 2017-03-07:
The following references can help you.

1. [user_runtime.h](https://github.com/f9micro/f9-kernel/blob/3f733494082c329d00e6824b609c5b62de032694/user/include/user_runtime.h)
2. [F9-Kernel Note](https://hackpad.com/F9-Kernel-Note-UnUXDVd9Zv2)
3. [f9-kernel](http://wiki.csie.ncku.edu.tw/embedded/f9-kernel)

gapry, 2017-03-07:
BTW, if you want to know more about microkernel, you can search the following references.

1. [cs9242(highly recommend)](https://www.cse.unsw.edu.au/~cs9242/16/lectures/)
2. [seL4 Homw Page](https://sel4.systems/)
3. [microkernel List](http://www.microkernel.info/)

gnitnaw, 2017-03-08:
Thanks. Finally I found the problem: I have to put the address where I have to touch in DECLARE_FPAGE.

One more question: Why it cannot run as well as lcd_test run?
I have to remove led(what I have added) or lcd_test(included in the repo.) from build.mk .
Is it normal?

grapherd, 2017-03-09:
@gnitnaw LCD_test have a `while (1)` loop at the end (line 35).

At this moment, F9 can't schedule user thread, so if someone makes an infinity loop (as LCD_test), you will get the result: hang it.

So, it is normal, but not a feature, is a bug which wait for fix.

gnitnaw, 2017-03-11:
When will you fix this bug?

grapherd, 2017-03-11:
@gnitnaw That's a big problem, ha. It needs to change the scheduler inside the user space, I have done some experiment in #129, but the result is not good as we want.

Maybe you can take this job, we want a pluggable scheduler for user space, and resolve this problem.

gnitnaw, 2017-03-11:
I'm happy to help you but until now I don't understand the way you do the context-switch...
(I tried to get more inform. form NCKU wiki, but still not so clear...)

gnitnaw, 2017-03-13:
Hi, 

If I understand well, the scheduler is done by kernel/sched.c
So I need to implement tcb_t *schedule_select() ?

gnitnaw, 2017-03-13:
Do you mind to show your implement for sched_rr and tell me where the problem is. I can try to improve it.

gapry, 2017-03-13:
UTCB is better. 

If your final goal is that F9 can provide  GPIO and I2C driver, I think you need to implement the following user apps.

1. GPIO and I2C driver
2. A genernal model object like kobject in Linux Kernel
3. Userspace Scheduler (use UTCB to do it)

Reference: 
1. [L4 Programming Introduction, Lecture 02, COMP9242 2003/S2](http://www.cse.unsw.edu.au/~cs9242/03/lectures/lect02.pdf)
2. [Microkernel Evolution:Designs and Aspects, Jim Huang, 2013](https://www.slideshare.net/jserv/microkernel-evolution)