f9micro/f9-kernel #154: Stack dump on qemu_stm32

xiangzhai, 2017-11-01:
Hi F9 developers,

```
$ qemu_stm32/arm-softmmu/qemu-system-arm -M stm32-p103 -kernel build/stm32p103/f9.bin -serial stdio -semihosting -s -S

(process:11600): GLib-WARNING **: gmem.c:483: custom memory allocation vtable not supported
VNC server running on `::1:5900'
LED Off
```

```
$ /data/download/arm-2014.05/bin/arm-none-linux-gnueabi-gdb 
GNU gdb (Sourcery CodeBench Lite 2014.05-29) 7.7.50.20140217-cvs
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-pc-linux-gnu --target=arm-none-linux-gnueabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://sourcery.mentor.com/GNUToolchain/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target remote :1234
Remote debugging using :1234
0x080043f8 in ?? ()
(gdb) bt
#0  0x080043f8 in ?? ()
#1  0x00000000 in ?? ()
(gdb) c
Continuing.
```

```
Reached end of schedule()
-------KTABLES------
-------KTIMER------

ktimer events: 
EVENT    DELTA       
-------NOW------
Now is 0
-------SOFTIRQ------
Kernel timer events              not scheduled
Asynchronous events              not scheduled
System calls                     not scheduled
KDB enters                       not scheduled
-------THREADS------
type  global   local    state  parent
-------MPU------
-------MEMPOOLS------
ID NAME                 SIZE       [START   :END     ] FLAGS     
 0 KTEXT                     21984 [08001000:080065e0] r-x --- N 
 1 UTEXT                      9472 [2000f200:20011700] --- r-x M 
 2 KIP                         512 [20000400:20000600] rw- r-- S 
 3 KDATA                       928 [20000600:200009a0] rw- --- N 
 4 KBSS                      59312 [20000a00:2000f1b0] rw- --- N 
 5 UDATA                       768 [20011700:20011a00] --- rw- M 
 6 UBSS                        256 [20011a00:20011b00] --- rw- M 
 7 MEM0                      42184 [20011b38:2001c000] --- rw- S 
 8 KBITMAP                      56 [20011b00:20011b38] rw- --- N 
 9 APB1DEV                   30720 [40000000:40007800] --- rw- D 
10 APB2_1DEV                 19456 [40010000:40014c00] --- rw- D 
11 APB2_2DEV                  3072 [40014000:40014c00] --- rw- D 
12 AHB1_1DEV                 15360 [40020000:40023c00] --- rw- D 
13 AHB1_2DEV                115712 [40023c00:40040000] --- rw- D 
14 AHB2DEV                  397312 [50000000:50061000] --- rw- D 
15 AHB3DEV              1073745920 [60000000:a0001000] --- rw- D 
-------AS------
-------TOP------
Init sampling...

Stack dump:
20000330 00000000 200013e0 00000006 00000000 200013d8 00000000 cccccccd 
0800424f 08005dbc 000020ac 0000202c 00000000 40004400 0000000c 20000618 
00000020 200007d7 0800174b fffffff9 00000001 000020ac 0000202c 40004400 
00003000 0800194b 08001550 01000200 00000000 0000000d 080018f5 08005e52 
00000000 08001681 0000000a 080021d5 00002000 200003e4 00000020 200003e8 
00000000 0000e7b0 00000000 00000000 00000000 00000000 00000000 00000000 
00000000 0800209f 080016cd 200003e4 0000e7b0 080043b9 08005e50 200007d8 
00000000 20001384 e000ed24 080044ad 00000000 00000000 00000000
```

```
$ arm-2014.05/bin/arm-none-linux-gnueabi-addr2line -p -i -e f9.elf -a 0x080043f8
0x080043f8: /data/project/xiangzhai/f9-kernel/kernel/start.c:76
```

But nothing wrong [kernel/start.c:76](https://github.com/f9micro/f9-kernel/blob/master/kernel/start.c#L76) ?

I use https://sourcery.mentor.com/GNUToolchain/package12813/public/arm-none-linux-gnueabi/arm-2014.05-29-arm-none-linux-gnueabi-i686-pc-linux-gnu.tar.bz2  to build f9. and this toolchain is able to build L4Re and run successfully hello testcase with qemu-system-arm http://os.inf.tu-dresden.de/pipermail/l4-hackers/2017/008151.html

Regards,
Leslie Zhai

mlouielu, 2017-11-01:
@xiangzhai In my far memory, I remember that stm32-p103 emulator isn't working, as you said in #127, it just doesn't work. I remember my development is always on stm32f429.

mlouielu, 2017-11-01:
I'm not sure if you are interested in [Piko-RT](https://github.com/piko-rt/pikoRT), another real-time kernel develop in the same team, it is under development now, and more active then f9.

xiangzhai, 2017-11-01:
@mlouielu Thanks for your kind response! I will try Piko-RT :)

closed 2017-11-17 by xiangzhai

slpbaby, 2019-03-06:
I know this is quite late, but I found the interrupt is triggered before kernel completes initialization. That's why hit the end of schedule error. For who that's might be interested, I have [commit a dirty fix](https://github.com/slpbaby/f9-kernel/commit/6eb8388545ac47482e77917c050e4649cf4901a6) to solve the issue, but currently only tested on qemu.